OpenVoiceOS/ZZZ-raspOVOS #6: mycroft.voice service deactivating

dickorydock, 2023-03-02:
When I look at the services, mycroft.service says active (exited) but I think thats OK? But then mycroft-voice.service doesn't look right:

 mycroft-voice.service - Mycroft Voice
     Loaded: loaded (/home/ovos/.config/systemd/user/mycroft-voice.service; enabled; vendor preset: enabled)
     Active: deactivating (stop-sigterm) since Thu 2023-03-02 18:11:56 EST; 13s ago
   Main PID: 792 (python)
      Tasks: 16 (limit: 1596)
        CPU: 19.077s
     CGroup: /user.slice/user-1000.slice/user@1000.service/app.slice/mycroft-voice.service
             ├─792 python /home/ovos/.local/bin/mycroft-systemd-voice
             └─800 /home/ovos/.local/share/precise03/precise-engine/precise-engine /home/ovos/.local/share/precise03models/>

Mar 02 18:11:56 raspberrypi mycroft-systemd-voice[792]: Traceback (most recent call last):
Mar 02 18:11:56 raspberrypi mycroft-systemd-voice[792]:   File "/home/ovos/.local/lib/python3.9/site-packages/mycroft/liste>
Mar 02 18:11:56 raspberrypi mycroft-systemd-voice[792]:     self.start_async()
Mar 02 18:11:56 raspberrypi mycroft-systemd-voice[792]:   File "/home/ovos/.local/lib/python3.9/site-packages/mycroft/liste>
Mar 02 18:11:56 raspberrypi mycroft-systemd-voice[792]:     self.stt = STTFactory.create()
Mar 02 18:11:56 raspberrypi mycroft-systemd-voice[792]:   File "/home/ovos/.local/lib/python3.9/site-packages/mycroft/liste>
Mar 02 18:11:56 raspberrypi mycroft-systemd-voice[792]:     return OVOSSTTFactory.create(config)
Mar 02 18:11:56 raspberrypi mycroft-systemd-voice[792]:   File "/home/ovos/.local/lib/python3.9/site-packages/ovos_plugin_m>
Mar 02 18:11:56 raspberrypi mycroft-systemd-voice[792]:     return clazz(plugin_config)
Mar 02 18:11:56 raspberrypi mycroft-systemd-voice[792]: TypeError: 'NoneType' object is not callable

builderjer, 2023-03-03:
do you have logs? It probably does not have a plugin installed.
create a file called `~/.config/mycroft/mycroft.conf`
add this to the file and save
```
{
  "log_level": "DEBUG"
}
```
restart mycroft
`systemctl --user restart mycroft`
let everything reboot and look at the file `~/.local/state/mycroft/voice.log`

dickorydock, 2023-03-03:
Thanks for showing me how to add a log! Here it is:

2023-03-03 00:01:18.257 - voice - mycroft.listener.mic:__init__:368 - INFO - Creating VAD engine: o>
2023-03-03 00:01:18.368 - voice - mycroft.listener.stt:create:12 - INFO - Creating STT engine: ovos>
2023-03-03 00:01:18.754 - voice - ovos_plugin_manager.utils:load_plugin:118 - WARNING - Could not f>
2023-03-03 00:01:18.758 - voice - ovos_plugin_manager.stt:create:137 - ERROR - The selected STT plu>
Traceback (most recent call last):

builderjer, 2023-03-03:
Good. The logs are great for debugging,  but I would suggest removing the log level when you are done debugging. It's a lot of writing to the SD card

closed 2023-03-03 by builderjer