OpenVoiceOS/ZZZ-raspOVOS #7: A little more information about cli-client?

dickorydock, 2023-03-03:
Since I'm still debugging why the mic doesn't respond, it would be helpful to add a note about how to actually access the cli-client in the section where you introduce that. 

I tried `./start-mycroft.sh cli` but got a bunch of errors

dickorydock, 2023-03-03:
Nevermind, I figured it out by changing my etc/default/locale file like they did in the answer to [this question](https://stackoverflow.com/questions/14547631/python-locale-error-unsupported-locale-setting)

builderjer, 2023-03-03:
Also, did you set the locale with `sudo raspi-config`

dickorydock, 2023-03-03:
Yes, I used `sudo raspi-config`.

Output from  `aplay -l`:
```
card 0: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 1: vc4hdmi [vc4-hdmi], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 3: UACDemoV10 [UACDemoV1.0], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

Output from `arecord -l`:
```
**** List of CAPTURE Hardware Devices ****
card 2: CameraB409241 [USB Camera-B4.09.24.1], device 0: USB Audio [USB Audio]
  Subdevices: 0/1
  Subdevice #0: subdevice #0
card 3: UACDemoV10 [UACDemoV1.0], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

```

builderjer, 2023-03-03:
and I am assuming that you want to use the `UACDemoV10` device for output, and the USB camera for a mic?
If you set those devices as active using `alsamixer` do you get any change?

dickorydock, 2023-03-03:
Here's what I tried:
`alsamixer`
F3 (Show playback controls)
F6 (Select sound card)
Selected `UACDemoV1.0`
Enter
F4 (Show capture controls)
F6 (Select sound card)
Selected `USB Camera-B4.09.24.1`
Enter
At this point it exits alsamixer and I get the error:
`cannot load mixer: Invalid argument`

builderjer, 2023-03-03:
Try a reboot and see if things load properly after that.  There may need to be a `.asoundrc` file according to this site
https://9to5answer.com/alsamixer-cannot-open-mixer-invalid-argument

dickorydock, 2023-03-03:
I tried reboot but that didn't change anything. I have followed the steps there but when I run `ovos@raspberrypi:/etc $ cat /proc/asound/cards
` I see 
``` 0 [Headphones     ]: bcm2835_headpho - bcm2835 Headphones
                      bcm2835 Headphones
 1 [vc4hdmi        ]: vc4-hdmi - vc4-hdmi
                      vc4-hdmi
 2 [CameraB409241  ]: USB-Audio - USB Camera-B4.09.24.1
                      OmniVision Technologies, Inc. USB Camera-B4.09.24.1 at usb-3f980000.usb-1.3, hi
 3 [UACDemoV10     ]: USB-Audio - UACDemoV1.0
                      Jieli Technology UACDemoV1.0 at usb-3f980000.usb-1.4, full speed
```

I tried changing the values in `asound.conf` to 2 or 3 but no difference so far.

dickorydock, 2023-03-03:
No, both say `pa_context_connect() failed: Connection refused`

builderjer, 2023-03-04:
Pulseaudio is not running
`systemctl --user restart pulseaudio`

builderjer, 2023-03-04:
you may also have to disable `suspend-on-idle` in default.pa also
`sudo sed -i -e "s/^load-module module-suspend-on-idle/#load-module module-suspend-on-idle/" /etc/pulse/default.pa`
`systemctl --user restart pulseaudio`

builderjer, 2023-03-14:
@dickorydock Has this been resolved?

builderjer, 2023-03-21:
I believe that these have been resolved in the dev branch.  I am closing this but feel free to open another with new information

closed 2023-03-21 by builderjer