OpenVoiceOS/ZZZ-raspOVOS #9: ERROR - [Errno2] No such file or directory: 'paplay'

dickorydock, 2023-03-03:
I get this error when running `ovos-cli-client.

I was able to fi`x it using `sudo apt-get install pulseaudio-utils` maybe add to the guide?

builderjer, 2023-03-03:
I am working on an updated version now.  Thanks for the insite

builderjer, 2023-03-04:
This install guide uses the latest code from git.  When you installed precise-lite, it used the version on PyPi which created some dependency issues.  
I would suggest using `--pre` in your pip command
`pip install --pre ovos-ww-plugin-precise-lite`

I am not sure how to fix this other than starting over

dickorydock, 2023-03-04:
Before I try that again with --pre, cna you clarify whether the wakeword is supposed to work with just the `ovos-core` from git?

builderjer, 2023-03-14:
@dickorydock has this been resolved?

builderjer, 2023-03-21:
I think all of these have been resolved in the dev branch.  I am closing this, but feel free to open another with new info

closed 2023-03-21 by builderjer