OpenVoiceOS/ZZZ-raspOVOS #11: Doesn't cover how to install skills

dickorydock, 2023-03-04:
The link in Section 6 sort of talks about how to develop skills but doesn't convey how to install them. Previous versions of this documentation mentioned how to install date-time for example, which I found helpful.

brandonne, 2023-03-10:
I think that very soon we can just put in a link to the user docs that will provide that information.

builderjer, 2023-03-11:
https://openvoiceos.github.io/community-docs/skill_install/

This will be put in the readme also

builderjer, 2023-03-14:
@dickorydock Now that there is a section in the docs about installing skills, can this be closed?

builderjer, 2023-03-20:
I am closing this as the [community docs](https://openvoiceos.github.io/community-docs/skill_install/) are pretty good, and I have had confirmation that OSM is working to install skills also