OpenVoiceOS/ZZZ-raspOVOS #27: Change "if" testing in manual_install.sh

brandonne, 2023-03-15:
IF you change your if statements from:
`if [[ !  $systemd || $systemd[0] == "y" || $systemd[0] == "Y" ]]; then`
to
`if [[ -n  $systemd || $systemd[0] == "y" || $systemd[0] == "Y" ]]; then`
(changing "!" to "-n")
They should properly validate and trigger the appropriate functions.  

It would also be nice if the "Type Y to start install also triggered on "y".

Also, there is currently no mycroft-admin-phal.service in `.config/systemd/user`

builderjer, 2023-03-15:
I don't get an error the way the 'if statements' are.  Is it not working for you? 
Also, I did add 'y' to the install

I am aware of the admin phal.  That will be a big change, but should be done this week
I have to work around my 'real job' also

builderjer, 2023-03-15:
The if statements had to check for an empty string.  Now it works as expected I think

brandonne, 2023-03-15:
They didn't work when I tested them.  They don't through an error, but it never ran the install_systemd function.  I added echo statements right before We are now ready to install OVOS and $systemd shows "y" instead of "YES" so it never triggers the function.

builderjer, 2023-03-15:
I just updated it this morning. Try that

brandonne, 2023-03-15:
Added to manual_install.sh:
```
echo
echo "We are now ready to install OVOS"
echo "systemd install" $systemd
echo " enabled" $enabled
echo "extra skills" $extra_skills
read -p "Type 'Y' to start install (any other key aborts): " install
if [[ $install != "Y" || $install != "y" ]]; then
```
Running the script shows that those variables don't get set to "YES" so the extra functions don't run:
```
ovos@raspberrypi:~/raspbian-ovos $ ./manual_install.sh 
This file will install ovos-core to this device
using the latest commits from github.

First lets setup some things.

Do you want to install systemd files (Y/n): y

Would you like to install extra skills to match the downloadable image? (Y/n): y

We are now ready to install OVOS
systemd install y
 enabled
extra skills y
Type 'Y' to start install (any other key aborts): 
```
Change the lines:
```
echo "First lets setup some things."
echo
read -p "Do you want to install systemd files (Y/n): " systemd
if [[ -n $systemd || $systemd[0] == "y" || $systemd[0] == "Y" ]]; then
    systemd="YES"
    echo
    read -p "Do you want to automatically start the ovos services? (Y/n): " enabled
    if [[ -n $enabled || $enabled[0] == "y" || $enabled[0] == "Y" ]]; then
        enabled="YES"
    fi
fi
echo
read -p "Would you like to install extra skills to match the downloadable image? (Y/n): " extra_skills
if [[ -n $extra_skills || $extra_skills[0] == "y" || $extra_skills[0] == "Y" ]]; then
    extra_skills="YES"
fi
echo
echo "We are now ready to install OVOS"
echo "systemd install" $systemd
echo " enabled" $enabled
echo "extra skills" $extra_skills
read -p "Type 'Y' to start install (any other key aborts): " install
if [[ $install != "Y" || $install != "y" ]]; then
    exit 0
```
Now we get expected behavior:
```
ovos@raspberrypi:~/raspbian-ovos $ ./manual_install.sh 
This file will install ovos-core to this device
using the latest commits from github.

First lets setup some things.

Do you want to install systemd files (Y/n): y

Do you want to automatically start the ovos services? (Y/n): y

Would you like to install extra skills to match the downloadable image? (Y/n): y

We are now ready to install OVOS
systemd install YES
 enabled YES
extra skills YES
Type 'Y' to start install (any other key aborts):
```

brandonne, 2023-03-15:
Figured it out.  brackets don't work that way:
```
read -p "Do you want to install systemd files (Y/n): " systemd
echo "full variable:" $systemd
echo "bracket:" $systemd[0]
echo "brace: ${systemd:0:1}
```
gives:
```
Do you want to install systemd files (Y/n): y
full variable: y
bracket: y[0]
brace: y
```
So new line is:
```
if [[ $systemd == "" || ${systemd:0:1} == "y" ||${systemd:0:1} == "Y" ]]; then
```

builderjer, 2023-03-15:
new try.  I am learning bash along the way

brandonne, 2023-03-16:
Yeah, I used to be really good at bash, but these days I use ansible for almost everything.

missed one of the conditionals, for enable ovos.
If we're going to leave in the debug code for echoing the variables, should put a label in front of them:
`echo $systemd`
to
`echo "Install Systemd files:" $systemd`

Also, line 202 if you put backticks inside of a quote it doesn't show the text.
I think we can close this issue, I have some suggestions but I'll open a separate issue for that.

builderjer, 2023-03-17:
closed with last fix.  thanks @brandonne for help in debugging

closed 2023-03-17 by builderjer