OpenVoiceOS/ZZZ-raspOVOS #42: Temperature incorrect when system_unit set to "english"

Bulwagga, 2023-04-04:
Running raspbian-ovos-dev.zip (4/3/2023) on a 3B+
After a clean install, I ran ovos-cli-client.
I asked "What is the weather"
and the answer was 13 degrees celsius (correct!)
Then I ran ovos-config set -k system_unit  "english"
and asked "What is the weather"
and the answer was 286 degrees.

builderjer, 2023-04-04:
Try `imperial ` instead of English. And actually,  this is a skill issue. I will move it if imperial does not work

Bulwagga, 2023-04-04:
"imperial" works, "english" does not.  I picked "english" out of some mycroft documentation, so you'll likely be faced with this again.  Could I suggest an enhancement? Extend the ovos-config dialog to replace the word "english" automatically with the word "imperial"? Or as a more general change, trap bad keywords and suggest alternatives?  Thank you, I am otherwise all set.

closed 2023-04-04 by Bulwagga

emphasize, 2023-04-04:
Still an issue in the skill as it requests 
```
"GET /v1/owm/onecall?exclude=minutely&lang=de&lat=49.5875&lon=8.35778&units=fahrenheit HTTP/1.1" 200 -
```
But none of your concern

emphasize, 2023-04-04:
btw also a problem with ovos-config. you shouldnt be able to pass "a-string-to-your-liking" to the config