OpenVoiceOS/ZZZ-raspOVOS #44: Warning message that language is defined twice

Bulwagga, 2023-04-05:
Running a RP3B+, fresh install of 
[raspbian-ovos-dev.zip](https://downloads.openvoiceos.com/images/raspbian/raspbian-ovos-dev.zip)                              04-Apr-2023 15:42          1327830572

When running ovos-cli-client, I repeatedly get the message:

server:_validate_args_combo:93 - WARNING - lang defined twice! choosing en_GB over en_US for voice: en_UK/apope_low

denics, 2023-05-07:
it is safe to ignore, but yes, it would be nicer to remove that from the default configuration I think

builderjer, 2023-05-07:
Mimic3 config sets a language, and it also pulls from mycroft.conf. in this case, mimic3 is using en_GB in the voice, and en_US from mycroft.conf. The next image will not have mimic3 as a default,  and the error should go away

builderjer, 2023-07-02:
I believe this has been resolved by using `ovos-tts-plugin-server`

closed 2023-07-02 by builderjer