OpenVoiceOS/ZZZ-raspOVOS #48: OVOS access point not created.  "Error: Getting access points failed

Bulwagga, 2023-05-07:
Running raspbian-ovos-headless.zip, 30-Apr-2023 23:26 on a Raspberry Pi 3B+.

When doing the first boot, I occasionally don't get an OVOS access point to set up my network login.  The message in syslog is

Dec 22 04:56:39 raspOvos ovos-systemd-phal[644]: 2022-12-22 04:56:39.575 - OVOS - ovos_PHAL_plugin_balena_wifi:display_n
etwork_setup:134 - INFO - Balena Wifi In Network Setup
Dec 22 04:56:39 raspOvos ovos-systemd-phal[644]: 2022-12-22 04:56:39.590 - OVOS - ovos_PHAL_plugin_balena_wifi:display_n
etwork_setup:137 - INFO - Spawning new wifi_process
Dec 22 04:56:40 raspOvos ovos-systemd-phal[644]: 2022-12-22 04:56:40.089 - OVOS - ovos_PHAL_plugin_balena_wifi:display_n
etwork_setup:198 - ERROR - #033[1;31mError: Getting access points failed#033[0m

I realized today that this only happens when the Pi is plugged in the power strip in our office (118.5V). When I'm working in the kitchen I get the OVOS access point much more frequently (120.1V). These happen to be the low and high values in our home, so I can't do much more experimentation.  There is no mention of low power in the logs.

There is a workaround, of course. I can put the network parameters directly onto the SD card immediately after writing.

builderjer, 2023-05-07:
This should only happen once. After the wifi has been connected,  it should not have to reconnect again.  I guess this is where I am getting lost. Why ate you having to reconnect to wifi just by moving your device? Or are you reinstalling an image? Or???

My questio. Is why is it trying g to create the AP again at all if the wifi you want to connect to is the same throughout your house.

Bulwagga, 2023-05-07:
I am making an environment setup script, leading me to perform the first 
raw installation each time I change my setup. So I have done perhaps 10 
image installs in the past week.  It works fine when I get the OVOS 
access point, but sometimes the AP doesn't come up and I get the error 
message.  It took me a while to realize the failure was correlated with 
which outlet I was using. Then I tested the voltage to see if that was 
different between the outlets, and it was.  That may or may not be the 
cause.

builderjer, 2023-05-07:
I can't see why that would be the cause. Your power adapter should more than take care of that. Unless there is an issue with it.  

Let's assume the power supply is good and the line voltage are not the problem 

You said you are comfortable modifying the image before you use it. Can you enable debugging and try again in those areas?

Edit/create `/home/ovos/.config/mycroft/mycroft.conf `

Put 

```
{
"log_level": "DEBUG"
}
```

And run you image with that enabled

Bulwagga, 2023-05-07:
Done. Which log files would you like to see?

builderjer, 2023-05-07:
The same as was shown before. Should be phal.log or ovos.log

Bulwagga, 2023-05-08:
No phal.log... "sudo find /media/myname/rootfs -name phal.log" returns nothing.
Searching files in /home/ovos/.local/state/mycroft for phal and balena turned up the failure in:
[ovos.log](https://github.com/OpenVoiceOS/raspbian-ovos/files/11416249/ovos.log)

and turned up mentions of "timeout" in:
[bus.log](https://github.com/OpenVoiceOS/raspbian-ovos/files/11416251/bus.log)

Let me know if you can't get to these files and I will post snippets.
Thank you for looking into this.

builderjer, 2023-05-08:
from reading the `ovos.log`, and this line
`2022-12-22 04:56:40.809 - OVOS - ovos_PHAL_plugin_balena_wifi:display_network_setup:198 - ERROR - [1;31mError: Getting access points failed[0m`
which repeats, leads me to believe that the pi can not find your AP on the wifi that it is trying to connect to.  I would say that your signal strength in that room is not good.

You can access it manually because it is looking for a specific AP, so even with a low signal, it connect to it.

This is my best guess.  Is your office, or the room you are having troubles in further away from your wifi router?

The timeout errors in `bus.log` are because it is looking for an AP and can't find one, so it restarts the process.

On another note, if you attach a keyboard and screen temporarily, `raspi-config` and `nmtui` are both available to use

Bulwagga, 2023-05-08:
I think you've put me onto the right issue.  Our neighbor (two houses away) has jacked his wifi power up high enough that his signal is stronger than the signal of the neighbor that lives between us. The room where the OVOS network fails to initialize has line of sight to the house with the super signal. I will play with this and update if I can nail down the issue any further.

It's important to note that when the initialization boot fails, the OVOS network choice never appears.  I never got to the stage of choosing an access point from a list. I interpreted the "Getting access points failed" message as meaning that the list of local access points was never generated. 

To further bury the low voltage idea, I checked the full logs for any indication of low voltage or frequency throttling and didn't find any difference between the two rooms.  Bogomips (whatever they really are) were also the same in both rooms.

Maybe this is TMI, but I find it really useful to work through scripting for system setup.  It's self documenting and super fast. The fact that you have set up ssh by default is a real win for my hobby time. With ssh, I can get access to raspi-config and nmtui in your raspbian environment from any room or any computer in the house. I think your concept of a headless raspbian image is poised to scale.

builderjer, 2023-05-08:
I'm glad that we can at least get this narrowed down. There has historically been issues with the balena wifi plugin, but no suitable replacement yet either. 

FYI, this is just a hobby for me too. I am a carpenter,  and it's spring/summer. Work picks up this time of year. I would gladly accept PR's for any improvements or bug fixes. And I will try to address any issues as quickly as I can.

builderjer, 2023-07-02:
I believe this is no longer an issue.  I am closing for now, and can be reopened, or a new issue made if needed.

closed 2023-07-02 by builderjer