OpenVoiceOS/ZZZ-raspOVOS #51: No internet radio.  With clean install: ERROR - The TTS plugin "ovos-tts-plugin-piper" could not be loaded.

Bulwagga, 2023-05-21:
After installing [raspbian-ovos-headless.zip](https://downloads.openvoiceos.com/images/raspbian/raspbian-ovos-headless.zip) on a 3B+, the audio log shows that "The TTS plugin "ovos-tts-plugin-piper" could not be loaded."  Somewhat later there is a message "ERROR - Audio service failed to launch (No TTS loaded)".

I load espeak phonemizer with 
sudo apt-get --assume-yes install libespeak-ng1 espeak-ng
and
pip install espeak-phonemizer

Then audio.log shows me "ERROR - The TTS plugin "ovos-tts-plugin-mimic" could not be loaded." and again the audio service fails to launch with (unkown plugin).

Then I add mimic with
pip install ovos-tts-plugin-mimic

What else do I need to do to hear internet radio?  I've tried several of the OCP skills and I'm having trouble getting my utterances recognized so I wrote a simple skill that gets me to the point of 

stream-player-skill:handle_stream_intent:40 - INFO - playing http://playerservices.streamtheworld.com/api/livestream-redirect/WAMCHD2.
mp3

but I'm still getting no sound.  Any suggestions?  mpg123 of the same link gives me the audio correctly, so at least some of my audio system is correct.

Thank you

Bulwagga, 2023-05-24:
After looking into the image build process, it looks to me like ovos-tts-plugin-mimic was already loaded during the system build. I will create another issue if I can find an earlier instance of any ovos-tts-plugin-mimic issues.

builderjer, 2023-05-24:
There are issues with missing 
espeak packages, i think. I don't remember off the top of my head, but they will be in the next build

Bulwagga, 2023-05-25:
No worries...  I just did a line-by-line through the first boot log from the first run of my most recent raspbian image.  No mention of issues with ovos-tts-plugin-mimic.  My sense is to let  this go until we build an image with piper & espeak packages and then see if it still has trouble finding mimic. I saw two other issues I will enter here.

builderjer, 2023-05-25:
That is normal when installing pip packages without sudo.

Bulwagga, 2023-05-25:
Good to know... Does this mean I should always be installing my ovos pip packages with sudo?

builderjer, 2023-05-26:
The user packages will take priority if they are installed. Most should be fine. Any packages that need sudo privileges should say so

builderjer, 2023-07-03:
I am going to close this as `ovos-tts-plugin-piper` loads now, but is unused for now.  The image defaults to `ovos-tts-server-plugin` with default public servers.

The other "internet radio" can be opened in another issue of its own

closed 2023-07-03 by builderjer