OpenVoiceOS/ZZZ-raspOVOS #53: Dinkum listener turns on and off repeatedly

Bulwagga, 2023-05-25:
Here is the first time dinkum-listener dies:
`ovos-systemd-skills[670]: 2022-12-22 04:58:11.724 - skills - ovos_utils.configuration:get_xdg_config_save_path:141 - WARNING - configuration moved to the `ovos_config`>
 ovos-systemd-skills[670]: 2022-12-22 04:58:11.756 - skills - mycroft.deprecated.skills.settings:__init__:99 - WARNING - skill_name is deprecated! use skill_id instead
 systemd[630]: ovos-dinkum-listener.service: start operation timed out. Terminating.
 systemd[630]: ovos-dinkum-listener.service: Failed with result 'timeout'.
 systemd[630]: Failed to start OVOS Dinkum Listener.
 systemd[630]: ovos-dinkum-listener.service: Consumed 10.413s CPU time.
 ovos-systemd-messagebus[648]: Uncaught exception GET /core (127.0.0.1)
 ovos-systemd-messagebus[648]: HTTPServerRequest(protocol='http', host='0.0.0.0:8181', method='GET', uri='/core', version='HTTP/1.1', remote_ip='127.0.0.1')
 ovos-systemd-messagebus[648]: Traceback (most recent call last):
 ovos-systemd-messagebus[648]:   File "/usr/local/lib/python3.9/dist-packages/tornado/websocket.py", line 630, in _run_callback
 ovos-systemd-messagebus[648]:     result = callback(*args, **kwargs)
 ovos-systemd-messagebus[648]:   File "/usr/local/lib/python3.9/dist-packages/ovos_messagebus/event_handler.py", line 52, in on_message
 ovos-systemd-messagebus[648]:     client.write_message(message)
 ovos-systemd-messagebus[648]:   File "/usr/local/lib/python3.9/dist-packages/tornado/websocket.py", line 331, in write_message
 ovos-systemd-messagebus[648]:     raise WebSocketClosedError()
 ovos-systemd-messagebus[648]: tornado.websocket.WebSocketClosedError
 ovos-systemd-messagebus[648]: Task exception was never retrieved
 ovos-systemd-messagebus[648]: future: <Task finished name='Task-3219' coro=<WebSocketProtocol13.write_message.<locals>.wrapper() done, defined at /usr/local/lib/python>
 ovos-systemd-messagebus[648]: Traceback (most recent call last):
 ovos-systemd-messagebus[648]:   File "/usr/local/lib/python3.9/dist-packages/tornado/websocket.py", line 1087, in wrapper
 ovos-systemd-messagebus[648]:     await fut
 ovos-systemd-messagebus[648]: tornado.iostream.StreamClosedError: Stream is closed
 ovos-systemd-messagebus[648]: During handling of the above exception, another exception occurred:
 ovos-systemd-messagebus[648]: Traceback (most recent call last):
 ovos-systemd-messagebus[648]:   File "/usr/local/lib/python3.9/dist-packages/tornado/websocket.py", line 1089, in wrapper
 ovos-systemd-messagebus[648]:     raise WebSocketClosedError()
 ovos-systemd-messagebus[648]: tornado.websocket.WebSocketClosedError
 ovos-systemd-messagebus[648]: Task exception was never retrieved
 ovos-systemd-messagebus[648]: future: <Task finished name='Task-3220' coro=<WebSocketProtocol13.write_message.<locals>.wrapper() done, defined at /usr/local/lib/python>
 ovos-systemd-messagebus[648]: Traceback (most recent call last):
 ovos-systemd-messagebus[648]:   File "/usr/local/lib/python3.9/dist-packages/tornado/websocket.py", line 1087, in wrapper
 ovos-systemd-messagebus[648]:     await fut
 ovos-systemd-messagebus[648]: tornado.iostream.StreamClosedError: Stream is closed
 ovos-systemd-messagebus[648]: During handling of the above exception, another exception occurred:
 ovos-systemd-messagebus[648]: Traceback (most recent call last):
 ovos-systemd-messagebus[648]:   File "/usr/local/lib/python3.9/dist-packages/tornado/websocket.py", line 1089, in wrapper
 ovos-systemd-messagebus[648]:     raise WebSocketClosedError()
 ovos-systemd-messagebus[648]: tornado.websocket.WebSocketClosedError
`
and I have attached a search through syslog of all the dinkum entries.

[dinkum.txt](https://github.com/OpenVoiceOS/raspbian-ovos/files/11560372/dinkum.txt)

I have problems with my microphone that may be related to the dinkum timeouts, but its not clear to me which problem is causing the which.

Bulwagga, 2023-05-25:
Doing my own build on May 25, 2023, I got the following messages about dependency conflicts (not exactly sure what this implies). I'm going to try see if I can update to the necessary packages and see if that helps the listener.

ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
ovos-core 0.0.8a21 requires adapt-parser<2.0.0,>=1.0.0, but you have adapt-parser 0.6.2 which is incompatible.
ovos-core 0.0.8a21 requires padacioso~=0.2, but you have padacioso 0.1.2 which is incompatible.
ovos-listener 0.0.2a10 requires ovos-backend-client>=0.0.7a2,~=0.0, but you have ovos-backend-client 0.0.6 which is incompatible.
ovos-listener 0.0.2a10 requires ovos-stt-plugin-vosk~=0.1, which is not installed.
ovos-listener 0.0.2a10 requires ovos-ww-plugin-precise>=0.1.1,~=0.1, which is not installed.
ovos-listener 0.0.2a10 requires ovos-ww-plugin-precise-lite>=0.1.2a3,~=0.1, which is not installed.

Bulwagga, 2023-05-25:
It turns out that the right padacioso was already installed in a different directory. If both versions are required we might need to specify the full paths.

ovos@raspOvos:~ $ pip uninstall padacioso
Found existing installation: padacioso 0.1.2
Uninstalling padacioso-0.1.2:
  Would remove:
    /home/ovos/.local/lib/python3.9/site-packages/padacioso-0.1.2.dist-info/*
    /home/ovos/.local/lib/python3.9/site-packages/padacioso/*
Proceed (Y/n)? y
  Successfully uninstalled padacioso-0.1.2
ovos@raspOvos:~ $ pip install padacioso
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: padacioso in /usr/local/lib/python3.9/dist-packages (0.2.1a4)
Requirement already satisfied: simplematch in /usr/local/lib/python3.9/dist-packages (from padacioso) (1.3)
ovos@raspOvos:~ $

builderjer, 2023-05-25:
I'm sorry I haven't had a chance to look at this yet. I will as soon as I can.

Bulwagga, 2023-05-25:
Dude... No hurry.  I enjoy debugging.  Springtime is when the hammers are working their hardest! So as I worked through the pyramid of dependencies, it turns out that obos-backend-client indirectly depends on something I didn't have in my build.  ovos-PHAL-plugin-color-scheme-manager. Judging by its name I am guessing that it has something to do with a user interface. A user interface that was probably not included in the headless build.

I added the color-scheme-manager and started a new build of ovos-listener.  Right now pip is trying to thread the needle across the dependency tree and shows no signs of stopping after 90 minutes. No hurry at this end. ;)

builderjer, 2023-05-26:
You can get past that dependency by using `pip install --no-deps <package info>`

builderjer, 2023-05-26:
And ovos-listener is only installed because I didn't delete it. I am trying to move to dinkum-listener

Bulwagga, 2023-05-26:
Aha! I have been using ovos-listener because it doesn't turn on and off like the dinkum-listener... I will remove ovos-listener from my build. Should simplify everything and force me to get the dinkum-listener working well. if this weird dependency hierarchy goes away I'll close this issue and open a new dinkum-listener issue thread that does NOT include ovos-listener. Thank you for your patience.

Bulwagga, 2023-05-26:
I have a achieved a clean build... Steps are

1. Install image [raspbian-ovos-headless.zip                        14-May-2023 21:33          1310642095
2. sudo pip uninstall ovos-listener
3. git clone https://github.com/OpenVoiceOS/raspbian-ovos.git
4. cd raspbian-ovos/
5. nano manual_install.sh
  - edit adapt-parser line to be: pip install adapt-parser~=1.0.0
  - edit padacioso line to be: pip install padacioso~=0.2
6. ./manual_install.sh

 I am closing this issue as the dependency problems (now resolved) are diluting the main issue.  I will open a new issue related to the dinkum listener if the problems still exist.

closed 2023-05-26 by Bulwagga

Bulwagga, 2023-05-26:
Yes. Error-message free after making those two edits.  Now I'm installing piper and espeak.  Trying to eliminate one issue at a time on the way to getting my microphone to work!  LOL.  I need to have a working microphone before I can be sure the dinkum-listener issue isn't being caused by lack of audio signal from the mic. I was using ovos-listener because on occasion it worked perfectly with my mic and didn't stop running until I killed the process. I understand that ovos-listener is even faster at responding to the wake word, so I'm stoked to get it running.