OpenVoiceOS/ZZZ-raspOVOS #55: Microphone not working

Bulwagga, 2023-05-26:
I'm using a 3B+ and have tested two devices. A $5 plugin USB mic, and a webcam with a microphone. The webcam has been used on another ubuntu PC, and both devices have worked successfully with ovos-listener.

The plugin works on this raspberry pi with arecord, parecord, and audacity. This problem may be related to or caused by the frequent restarting of Dinkum-listener noted here in another issue.

When using ovos-cli-client there is no output at all below the "Mic Level" text. When using ovos-listener this meter worked.

Searching syslog for pulse gives:
06:55:59 raspOvos pulseaudio[647]: Failed to find a working profile.
06:55:59 raspOvos pulseaudio[647]: Failed to load module "module-alsa-card" (argument: "device_id="2" name="platform-3f902000.hdmi" card_name="alsa_card.platform-3f902000.hdmi" namereg_fail=false tsched=no fixed_latency_range=no ignore_dB=no deferred_volume=yes use_ucm=yes avoid_resampling=no card_properties="module-udev-detect.discovered=1""): initialization failed.
06:56:41 raspOvos pulseaudio[647]: ALSA woke us up to read new data from the device, but there was actually nothing to read.
06:56:41 raspOvos pulseaudio[647]: Most likely this is a bug in the ALSA driver 'snd_usb_audio'. Please report this issue to the ALSA developers.
22 06:56:41 raspOvos pulseaudio[647]: We were woken up with POLLIN set -- however a subsequent snd_pcm_avail() returned 0 or another value < min_avail.
06:56:42 raspOvos ovos-systemd-phal[684]: 2022-12-22 06:56:42.288 - OVOS - ovos_PHAL.service:load_plugins:73 - INFO - PHAL plugin loaded: ovos-PHAL-plugin-pulseaudio

ovos@raspOvos:~ $ pactl info gives what I believe are the correct defaults:
Server String: /run/user/1000/pulse/native
Library Protocol Version: 34
Server Protocol Version: 34
Is Local: yes
Client Index: 96
Tile Size: 65472
User Name: ovos
Host Name: raspOvos
Server Name: pulseaudio
Server Version: 14.2
Default Sample Specification: s16le 2ch 44100Hz
Default Channel Map: front-left,front-right
Default Sink: alsa_output.platform-bcm2835_audio.analog-stereo
Default Source: alsa_input.usb-C-Media_Electronics_Inc._USB_PnP_Sound_Device-00.mono-fallback
Cookie: 664b:61dc

ovos@raspOvos:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: Device [USB PnP Sound Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

tried with and without an .asoundrc
ovos@raspOvos:~ $ cat .asoundrc
pcm.!default {
  type asym
  capture.pcm "mic"
}
pcm.mic {
  type plug
  slave {
    pcm "hw:[1],[0]"
  }
}

builderjer, 2023-05-27:
I know the mic meter is not implemented with dunkum-listener.

You may need one of the new mic plug-ins such as https://github.com/OpenVoiceOS/ovos-microphone-plugin-alsa

Bulwagga, 2023-05-27:
This solved it. Thank you!

Bulwagga, 2023-05-27:
Interestingly, once the system captures "hey mycroft" it appears to stop dinkum-listener.  I'm closing this and will add details as required to the dinkum-listener issue.

closed 2023-05-27 by Bulwagga

builderjer, 2023-05-27:
I think I sW something reported in the chats about that. I will add this to the image also