OpenVoiceOS/ZZZ-raspOVOS #56: Clean raspbian-lite install. Failed to start ovos.service: Unit ovos.service not found.

Bulwagga, 2023-05-27:
This may well be an oversight on my part, but I think I followed the instructions.

Following documentation at https://github.com/OpenVoiceOS/ovos-technical-manual/blob/master/docs/raspbian_ovos.md using a 3B+

Downloaded and modified the latest raspbian-lite image (this step is where I believe the user error happened). Then
   sudo apt install build-essential python3-dev python3-pip swig libssl-dev libfann-dev portaudio19-dev libpulse-dev cmake git libncurses-dev pulseaudio-utils
   cd
   git clone https://github.com/OpenVoiceOS/raspbian-ovos.git
   cd raspbian-ovos
   ./manual_install.sh 

There was a repeated warning message:
  WARNING: The script * is installed in '/home/ovos/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no->

The script ended with ovos failing to start:
Failed to start ovos.service: Unit ovos.service not found.

"systemctl --user start ovos" also produces the ovos.service not found message

builderjer, 2023-05-28:
To be honest, that script  was a hack to test with. It will ONLY work on the dev branch,  and I'm not sure even then. There have been package changes since I wrote this. For instance,  it does not contain any reference to dinkun-listener. I think the script itself is handy to have, and an update should be done with new packages ans systemd files. PR's are more than welcome,  and should be as simple as changing/addind/removing a few lines.

Bulwagga, 2023-05-29:
LOL! Well, it looked pretty good to me. Given the docs, I had thought it was the script you use to create the build image. I *was* puzzled by the lack of dinkum-listener but figured it was just one of the many things I don't understand yet.  As for PR's, I'll work towards that but it may be have to wait until I can generate a useful suggestion. :)

closed 2023-05-29 by Bulwagga