OpenVoiceOS/ZZZ-raspOVOS #58: respeaker install

JarbasAl, 2023-06-13:
perhaps helper scripts can be added to configure different boards and companion PHAL plugins. 

I recently setup a respeaker 2 mic in vanilla raspbian as follows

[image]

## /boot/config.txt

disabled onboard soundcard and hdmi audio and decreased GPU RAM as well as enabling wm8960 for respeaker 2 mic

changes
```
# Enable audio (loads snd_bcm2835)
# dtparam=audio=on

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d,noaudio

# setup less memory for GPU
gpu_mem=16

# enable respeaker 2 mic hat
# disable the onboard audio jack/hdmi sound by commenting (add #) above in line dtparam=audio=on
dtparam=i2s=on
dtoverlay=wm8960-soundcard
```

## alsa

the following worked for me

```bash
amixer cset numid=1 34,34
amixer cset numid=26 3
amixer cset numid=27 4
amixer cset numid=30 5
amixer cset numid=32 5
amixer cset numid=33 5
amixer cset numid=34 25
amixer cset numid=35 on
amixer cset numid=9 3
amixer cset numid=8 3
amixer cset numid=49 on
amixer cset numid=51 on
amixer cset numid=37 0
amixer cset numid=38 0
amixer cset numid=39 5
amixer cset numid=48 on
amixer cset numid=50 on
amixer cset numid=54 on
amixer cset numid=16 5
amixer cset numid=15 4
alsactl store
```

j1nx, 2023-12-11:
Implemented this here;
https://github.com/OpenVoiceOS/ovos-buildroot/commit/4acfa1202cf2262b8a0417ed6d56afa52fb9e61e

Will test this the coming days and tweak if needed. Will report back my findings.

builderjer, 2024-01-19:
I don't think this is still valid.  The newest images support the respeaker out of the box.  @JarbasAl  please reopen if needed

closed 2024-01-19 by builderjer