OpenVoiceOS/ZZZ-raspOVOS #59: Configurator enhancement: Adding installation options to install script

Bulwagga, 2023-06-17:
Should we add "configurator" options to simplify life for people assembling the system?
Consider padatious. It is optional and disabled by default, but users need to enable it in config for latest ovos-core. Should we offer the option of using the latest ovos-core, and then have the script add padatious automatically?  Why wouldn't the new ovos-core pull in the padatious it needed?

Could we offer other optional components? Should we update the local mycroft.conf automatically based on the chosen options?

There is product configurator software out there (https://github.com/OCA/product-configurator, for example) but seems like high overhead for a system that only has hierarchical dependencies.

builderjer, 2023-06-24:
The idea eventually is to have auto-detection of miscellaneous hardware such as the respeaker mics, or the aiy voicebonnet, or ... on the images.

But I don't think there needs to be options in the `manual_user_install.sh` script.  If they are installing this way, I would say they have the knowledge to setup their hardware also.

Unless you are talking about something else other than hardware

Bulwagga, 2023-06-24:
I'm thinking about an on-ramp for new users.  The thread that led me to enter this was about clusters of core skills that worked together better than others.  Strictly software.  On the face of it, every plugin should know enough to install it's own dependencies. Yet does every user know enough to include a tts plugin? Do some pairings of plugins work better than others?

It looks like (and I could be very wrong here) that certain plugins require users to make choices about modifications in mycroft.conf. No doubt these choices are documented, but it might be a long time before users work through the docs.

A good example might be the imperial vs. metric discussion. The install script might be a good place to get that decided. Something like a raspi-config?

Anyway, too early to do anything along these lines but I can see a day where we will be overwhelmed with user questions.

builderjer, 2023-06-24:
look at https://github.com/OpenVoiceOS/ovos-config
it has configuration abilities, with the command line

But what you want maybe more belong in [our docs](https://openvoiceos.github.io/community-docs/) and [github](https://github.com/OpenVoiceOS/community-docs)

builderjer, 2023-07-02:
I think I will close this one also.  maybe create and issue in the [community-docs](https://github.com/OpenVoiceOS/community-docs/issues)

closed 2023-07-02 by builderjer

Bulwagga, 2023-07-02:
Agree, let's wait and see how the user base develops.