OpenVoiceOS/ZZZ-raspOVOS #70: Setup help.

Iconology, 2023-07-20:
Hey all,
I just found this project the other day after searching for an alternative to Mycroft due to the halt in development. I was running their PiCroft OS on a Raspberry Pi 3B. You boot it up, it tells you the activation code, and you're good to go. I started off trying the Buildroot edition, but the first boot just got stuck at the main logo page with a spinning circle in the top corner. I let it sit for a few hours, but never moved. So, I opted to install Raspbian ovos, I'm a bit stuck on how to set it up. If I say the wake word, Hey Mycroft, it gives the audio clue that it heard the wake word, but there are no TTS replies. I'm sure I'm missing something; I just don't entirely know what.

Apologies for the noob questions; any help would be much appreciated. 
Thanks

Bulwagga, 2023-07-20:
Thank you for using raspovos. This is a development effort.  No worries about noob quesions.  I'm still asking them myself. Did you install one of the [images](https://ovosimages.ziggyai.online/raspbian/) or the [manual_user_install](https://github.com/OpenVoiceOS/raspbian-ovos)? We're chasing a similar issue now and it may be that you are seeing the same. Somewhere on your system would be a file called audio.log (either in /ramdisk/mycroft/audio.log or /home/ovos/.local/state/mycroft/audio.log.  If you could upload this file (drag and drop into your reply window) we could check.  As a test, please

reboot the pi (to clear out the old log files)
wait a minute or two for it to boot up fully
say "hey mycroft" and at the audio clue say "What time is it"
Wait another minute or two and then
say "hey mycroft" and at the audio clue say "What day is it"

Wait a minute or to and then make a copy of the audio.log file to upload. The bug is that the audio.log file will look normal but ovos won't be speak the answer.  Occasionally OVOS does start speaking, but at least on my system that's not fully under my control. :)

Thank you!

builderjer, 2023-07-20:
Yes, logs would be good. And did you use the image? Or the install script 

On a side note, you DO NOT have to wait for the audio cue to start asking questions. In fact, it may stop recording by the time you hear it. Not sure why there is a lag sometimes.

Iconology, 2023-07-20:
Thank you for the positivity and replies! Here is the output of audio.log. I believe it's currently running raspbian-ovos-dev_1689774023
I've tried the following images.
raspbian-ovos-dev_1688509994 | raspbian-ovos-dev_1689774023 |  manual_user_install

```
2023-07-20 12:15:43.510 - audio - ovos_utils.signal:get_ipc_directory:26 - WARNING - Deprecation version=0.1.0. Caller=ovos_audio.__main__:26. Expected a dict config and got None.
2023-07-20 12:15:43.654 - audio - ovos_audio.service:__init__:48 - INFO - Starting Audio Service
2023-07-20 12:15:43.662 - audio - ovos_audio.service:on_started:31 - INFO - Audio service started.
2023-07-20 12:15:43.753 - audio - ovos_bus_client.session:reset_default_session:531 - INFO - New Default Session Start: eaff635f-f65b-4991-95b3-b54f41220f57
2023-07-20 12:15:44.006 - audio - ovos_audio.service:_maybe_reload_tts:289 - INFO - (re)loading TTS engine
2023-07-20 12:15:44.791 - audio - ovos_utils.configuration:get_xdg_base - WARNING - Deprecation version=0.1.0. Caller=ovos_tts_plugin_mimic:93. configuration moved to the `ovos_config` package.
2023-07-20 12:15:44.960 - audio - ovos_utils.configuration:read_mycroft_config - WARNING - Deprecation version=0.1.0. Caller=ovos_tts_plugin_mimic:106. configuration moved to the `ovos_config` package.
2023-07-20 12:15:44.973 - audio - ovos_plugin_manager.tts:create:197 - INFO - Found plugin ovos-tts-plugin-server
2023-07-20 12:15:44.991 - audio - ovos_utils.file_utils:resolve_resource_file:152 - WARNING - Deprecation version=0.1.0. Caller=ovos_plugin_manager.templates.tts:526. Expected a dict config and got None.
2023-07-20 12:15:45.208 - audio - mycroft:<module>:37 - WARNING - mycroft has been deprecated! please start importing from ovos_core and companion packages
mycroft module remains available for backwards compatibility and will be removed in version 0.2.0
2023-07-20 12:15:52.363 - audio - ovos_plugin_manager.tts:create:200 - INFO - Loaded plugin ovos-tts-plugin-server
2023-07-20 12:15:52.371 - audio - ovos_audio.service:_maybe_reload_tts:308 - INFO - (re)loading fallback TTS engine
2023-07-20 12:15:52.631 - audio - ovos_plugin_manager.tts:create:197 - INFO - Found plugin ovos-tts-plugin-mimic
2023-07-20 12:15:52.640 - audio - ovos_utils.file_utils:resolve_resource_file:152 - WARNING - Deprecation version=0.1.0. Caller=ovos_plugin_manager.templates.tts:526. Expected a dict config and got None.
2023-07-20 12:15:52.993 - audio - ovos_utils.configuration:get_xdg_base - WARNING - Deprecation version=0.1.0. Caller=ovos_tts_plugin_mimic:93. configuration moved to the `ovos_config` package.
2023-07-20 12:15:53.003 - audio - ovos_utils.configuration:read_mycroft_config - WARNING - Deprecation version=0.1.0. Caller=ovos_tts_plugin_mimic:106. configuration moved to the `ovos_config` package.
2023-07-20 12:15:53.009 - audio - ovos_plugin_manager.tts:create:200 - INFO - Loaded plugin ovos-tts-plugin-mimic
2023-07-20 12:15:53.237 - audio - ovos_plugin_manager.audio:load_audio_service_plugins:84 - INFO - Loading audio service plugin: ovos_audio_simple
2023-07-20 12:15:53.242 - audio - ovos_plugin_manager.audio:load_audio_service_plugins:84 - INFO - Loading audio service plugin: ovos_common_play
2023-07-20 12:15:57.390 - ovos.common_play - WARNING - Tried to initialize ovos.common_play multiple times, ignoring
2023-07-20 12:15:59.370 - ovos.common_play - WARNING - Tried to initialize ovos.common_play multiple times, ignoring
2023-07-20 12:16:00.717 - audio - ovos_audio.audio:load_services:75 - INFO - OCP - OVOS Common Play set as default backend
2023-07-20 12:16:00.724 - audio - ovos_audio.service:on_alive:27 - INFO - Audio service is alive.
2023-07-20 12:16:11.678 - audio - ovos_plugin_common_play.ocp:register_ocp_intents:120 - INFO - OCP intents missing, registering for <ovos_plugin_common_play.ocp.OCP object at 0x7f54fb9c40>
2023-07-20 12:17:27.781 - audio - ovos_audio.service:execute_tts:320 - INFO - Speak: I don't know what that means.
2023-07-20 12:18:46.895 - audio - ovos_audio.service:execute_tts:320 - INFO - Speak: It's one eighteen
2023-07-20 12:24:59.727 - audio - ovos_audio.service:execute_tts:320 - INFO - Speak: It's one twenty four
2023-07-20 14:23:25.965 - audio - ovos_audio.service:execute_tts:320 - INFO - Speak: Currently three twenty three
2023-07-20 14:29:04.501 - audio - ovos_audio.service:execute_tts:320 - INFO - Speak: three twenty nine
2023-07-20 14:29:43.114 - audio - ovos_audio.service:execute_tts:320 - INFO - Speak: thursday, july twentieth, twenty twenty three
```

builderjer, 2023-07-20:
I would say there os something wrong with pulseaudio not switching to your speaker. Even with the online tts not working it should default to mimic 1. 

What is the output from
`pactl info`
And
`pactl  list sinks`

Bulwagga, 2023-07-20:
Is it valid to test pulseaudio by using ovos-speak, something like:
    ovos-speak hello
Or are there two separate mechanisms for tts depending on the source of the text?

builderjer, 2023-07-21:
Does it work with `ovos-speak "hello world"`

Bulwagga, 2023-07-21:
I had a successful query this morning and I realized it had the same unusual statement as the working trial that happened a few days ago (in italics):

2023-07-20 09:10:09.038 - audio - ovos_plugin_common_play.ocp:register_ocp_intents:119 - INFO - OCP intents missing, registering for <ovos_plugin_common_play.ocp.OCP object at 0x7f709c5ac0>
2023-07-20 09:53:11.216 - audio - ovos_plugin_common_play.ocp.player:play_next:483 - _ERROR - self.active_backend is undefined_
2023-07-20 09:53:11.480 - audio - ovos_plugin_common_play.ocp.player:play_next:517 - INFO - requested next, but there aren't any more tracks
2023-07-20 11:35:52.420 - audio - ovos_audio.service:execute_tts:320 - INFO - Speak: one thirty five

I put some debugging in ocp/player.py (where this message originates) but nothing has triggered it out of about 10 tries. So I can't "follow" it and find out what's going on.  Is the error message "self.active_backend is undefined" part of the software flow driving us to use mimic 1?  Odd that an error message is related to a successful outcome.

Iconology, 2023-07-21:
Here's the output for the two commands. ovos-speak works, audio goes through the 3.5mm speakers. I could be setting the defaults for pulse incorrectly, not quite sure.
Thanks!

> I would say there os something wrong with pulseaudio not switching to your speaker. Even with the online tts not working it should default to mimic 1.
> 
> What is the output from `pactl info` And `pactl list sinks`

pactl info
```
Server String: /run/user/1000/pulse/native
Library Protocol Version: 34
Server Protocol Version: 34
Is Local: yes
Client Index: 23
Tile Size: 65472
User Name: ovos
Host Name: raspOvos
Server Name: pulseaudio
Server Version: 14.2
Default Sample Specification: s16le 2ch 44100Hz
Default Channel Map: front-left,front-right
Default Sink: alsa_output.platform-bcm2835_audio.analog-stereo
Default Source: alsa_input.usb-OmniVision_Technologies__Inc._USB_Camera-B4.09.24.1-01.multichannel-input
Cookie: 6864:b4eb

```
pactl list sinks
```
Sink #0
        State: IDLE
        Name: alsa_output.platform-bcm2835_audio.analog-stereo
        Description: Built-in Audio Analog Stereo
        Driver: module-alsa-card.c
        Sample Specification: s16le 2ch 48000Hz
        Channel Map: front-left,front-right
        Owner Module: 6
        Mute: no
        Volume: front-left: 32768 /  50% / -18.06 dB,   front-right: 32768 /  50% / -18.06 dB
                balance 0.00
        Base Volume: 56210 /  86% / -4.00 dB
        Monitor Source: alsa_output.platform-bcm2835_audio.analog-stereo.monitor
        Latency: 54257 usec, configured 59954 usec
        Flags: HARDWARE HW_MUTE_CTRL HW_VOLUME_CTRL DECIBEL_VOLUME LATENCY
        Properties:
                alsa.resolution_bits = "16"
                device.api = "alsa"
                device.class = "sound"
                alsa.class = "generic"
                alsa.subclass = "generic-mix"
                alsa.name = "bcm2835 Headphones"
                alsa.id = "bcm2835 Headphones"
                alsa.subdevice = "0"
                alsa.subdevice_name = "subdevice #0"
                alsa.device = "0"
                alsa.card = "0"
                alsa.card_name = "bcm2835 Headphones"
                alsa.long_card_name = "bcm2835 Headphones"
                alsa.driver_name = "snd_bcm2835"
                device.bus_path = "platform-bcm2835_audio"
                sysfs.path = "/devices/platform/soc/3f00b840.mailbox/bcm2835_audio/sound/card0"
                device.form_factor = "internal"
                device.string = "hw:0"
                device.buffering.buffer_size = "10576"
                device.buffering.fragment_size = "2640"
                device.access_mode = "mmap"
                device.profile.name = "analog-stereo"
                device.profile.description = "Analog Stereo"
                device.description = "Built-in Audio Analog Stereo"
                module-udev-detect.discovered = "1"
                device.icon_name = "audio-card"
        Ports:
                analog-output: Analog Output (type: Analog, priority: 9900, availability unknown)
        Active Port: analog-output
        Formats:
                pcm

Sink #1
        State: IDLE
        Name: alsa_output.platform-3f902000.hdmi.hdmi-stereo
        Description: Built-in Audio Digital Stereo (HDMI)
        Driver: module-alsa-card.c
        Sample Specification: s16le 2ch 44100Hz
        Channel Map: front-left,front-right
        Owner Module: 7
        Mute: no
        Volume: front-left: 65536 / 100% / 0.00 dB,   front-right: 65536 / 100% / 0.00 dB
                balance 0.00
        Base Volume: 65536 / 100% / 0.00 dB
        Monitor Source: alsa_output.platform-3f902000.hdmi.hdmi-stereo.monitor
        Latency: 59804 usec, configured 59954 usec
        Flags: HARDWARE DECIBEL_VOLUME LATENCY SET_FORMATS
        Properties:
                alsa.resolution_bits = "16"
                device.api = "alsa"
                device.class = "sound"
                alsa.class = "generic"
                alsa.subclass = "generic-mix"
                alsa.name = "MAI PCM i2s-hifi-0"
                alsa.id = "MAI PCM i2s-hifi-0"
                alsa.subdevice = "0"
                alsa.subdevice_name = "subdevice #0"
                alsa.device = "0"
                alsa.card = "1"
                alsa.card_name = "vc4-hdmi"
                alsa.long_card_name = "vc4-hdmi"
                alsa.driver_name = "vc4"
                device.bus_path = "platform-3f902000.hdmi"
                sysfs.path = "/devices/platform/soc/3f902000.hdmi/sound/card1"
                device.form_factor = "internal"
                device.string = "hdmi:1"
                device.buffering.buffer_size = "10576"
                device.buffering.fragment_size = "2644"
                device.access_mode = "mmap"
                device.profile.name = "hdmi-stereo"
                device.profile.description = "Digital Stereo (HDMI)"
                device.description = "Built-in Audio Digital Stereo (HDMI)"
                module-udev-detect.discovered = "1"
                device.icon_name = "audio-card"
        Ports:
                hdmi-output-0: HDMI / DisplayPort (type: HDMI, priority: 5900, availability unknown)
        Active Port: hdmi-output-0
        Formats:
                pcm
```

Bulwagga, 2023-07-22:
These are the same outputs I'm seeing from the manual_install and 1689774023. The one difference is that when I install 1688509994, the audio works.

Iconology, 2023-07-22:
> These are the same outputs I'm seeing from the manual_install and 1689774023. The one difference is that when I install 1688509994, the audio works.

Thank you! Flashed back to the 1688509994 version and TTS audio is working. There's about a 20-30 second delay before the answer is spoken, but that's more of a minor inconvenience. Picroft was pretty snappy, and I'm sure in newer releases that'll be tweaked.

Do you happen to know where-is/how to edit the config file for changing the weather to Celsius/Fahrenheit and changing the TTS to Google? I installed it; I'm just not quite sure where to change the TTS module; I'm too used to MyCroft. 

Thank you both very much.

Bulwagga, 2023-07-22:
Sounds like the weather command is working. :) 

After a reboot, and from a command line logged in as ovos....

ovos-config set -k system_unit -v imperial

This will edit /home/ovos/.config/mycroft/mycroft.conf.

Have fun!

Bulwagga, 2023-07-22:
Seems to be the same as #69.

builderjer, 2023-07-22:
Im not sure what changed in those two versions.  I renamed a file and just switched it back to a simple form.  An image is building to test with this change

Iconology, 2023-07-22:
> Sounds like the weather command is working. :)
> 
> After a reboot, and from a command line logged in as ovos....
> 
> ovos-config set -k system_unit -v imperial
> 
> This will edit /home/ovos/.config/mycroft/mycroft.conf.
> 
> Have fun!

Thank you so much!

> Im not sure what changed in those two versions. I renamed a file and just switched it back to a simple form. An image is building to test with this change

I flashed the latest version (1690040653) on an extra sd card. After the first answer, audio stops working. Version (1688509994) works fine with replying. If that's of any help.

Bulwagga, 2023-07-22:
Very curious. The behavior is back to what it was when I first entered #69. @builderjer it may not be something we did. The manual_build also stopped working during the same time and I believe the only files of yours that I use are mycroft.conf and the ovos sytsemd files. The rest are downloaded directly from git. While you look at the images I'll see If I can find what else might have changed in git.

builderjer, 2023-07-24:
There has been a discussion on [Matrix support channel](https://matrix.to/#/!QMaxYxzZZsZPCUgTZw:matrix.org/$7yPaXaKbJ1dGKbLvly1llH8hueWpl8Z_ZGPhUk9Wbmg?via=matrix.org&via=kde.org&via=mozilla.org)

Don't think it is this repo

Bulwagga, 2023-07-24:
Thanks. I'm tracking it.  I am trying mimic first thing tomorrow morning and will post the result to matrix.+}

Bulwagga, 2023-07-24:
If you've read this far, there is a workaround for this... In $HOME/.config/mycroft/mycroft.conf specify a LOCAL text-to-speech engine... For example
    "tts": {
        "module": "ovos-tts-plugin-piper",
        "ovos-tts-plugin-piper": {},
        "fallback_module": "ovos-tts-plugin-mimic",
        "ovos-tts-plugin-mimic": {
            "voice": "ap"
            }
        }

builderjer, 2023-09-03:
I think all of this has been fixed with the new image and plugins.

I will close it, but feel free to open another if more problems arise

closed 2023-09-03 by builderjer