OpenVoiceOS/ZZZ-raspOVOS #71: Community documentation link in Readme.md is dead

Bulwagga, 2023-07-26:
"Check out the [community documentation](https://openvoiceos.github.io/community-docs/install_raspbian/) for information on how to install manually or build an image yourself."  install_raspbian.md has been moved to 'unused'.  Not sure whether we should delete the reference, move link to 'unused', or move install_raspbian.md back to 'docs'.

builderjer, 2023-07-26:
I think a section in the docs could be made to describe your manual-install script.  There really isn't one there now. 

Then the Readme here needs fixes also and get rid of the reference to build an image. Or add it to technical docs.

I'm on my phone now, but will do some searching for where things should be

builderjer, 2023-07-28:
I fixed the links for both the downloads, and the docs.  We should start a discussion on where this image and install script is going.

Bulwagga, 2023-07-28:
Thanks.  Sorry I haven't been discussing.  In general I buy into everything you've been suggesting.  Where is the right place to start and maintain a list of prioritized things to do? Today I will be putting in a fix to ovos-phal-admin service (it wasn't starting) as well as ovos_start.sh and ovos_stop.sh. I have also done some preliminary experiments with the install script on a ubuntu system to see what the issues might be. My 2 cents is that the manual install script allows people to easily install OVOS on existing systems.  I have four pi's scattered around the house that each do multiple things already.  It is most convenient for me to use the script to overlay OVOS. Same with my ubuntu systems. On the other hand, if I were developing code I would want to use an image to be confident about having a common starting point across developers. The images are also super useful to people that just want to take a test-drive.  You also mentioned having both dev and stable versions of the install script. Makes total sense.

builderjer, 2023-07-28:
Ha!! I didn't mean a discussion here between us, you do just fine. I would like to start a public discussion with maybe a roadmap and input from others

Let's start one here. https://github.com/orgs/OpenVoiceOS/discussions

builderjer, 2023-07-28:
Let's put it all in the discussions area. A checklist can be made, And each task just checked off. That way, hopefully we can have More community input.

Bulwagga, 2023-07-28:
Cool. You lead, I will follow. Thank you!

builderjer, 2023-07-28:
I will start something this evening after work

builderjer, 2023-07-30:
Going to close this and move it to [this discussion](https://github.com/orgs/OpenVoiceOS/discussions/185). 

The original issue with the link to the docs has been fixed with [this commit](https://github.com/OpenVoiceOS/raspbian-ovos/commit/58717324c533615c13294c6551913a7a00a83d80)

closed 2023-07-30 by builderjer