OpenVoiceOS/ZZZ-raspOVOS #85: [ENHANCEMENT] Mark 1 image

mikejgray, 2023-09-02:
I made quite a few customizations on the Mark 1 image to dial it in. Some are probably more personal preference than anything, but others are notes from Jarbas and some of my own. Please feel free to not include certain things with the image, like some of the skills, but I figured they helped give it some of the personality of the original.

- Swapped out Pi 3 for Pi 4 (not relevant to image build, but giving a little context)

- Update /boot/config.txt to allow USB boot (`program_usb_boot_mode=1`)
- Update /boot/config.txt to append: `dtoverlay=vc4-kms-v3d` -> `dtoverlay=vc4-kms-v3d,audio=off` (turns off HDMI audio, to limit potential audio confusion)
- `pip install padatious`
- `pip install neon-skill-speak`
- `pip install git+https://github.com/jarbasskills/skill-icanhazdadjokes`
- `pip install git+https://github.com/OpenVoiceOS/ovos-skill-easter-eggs`
- `pip install --force-reinstall --no-deps git+https://github.com/OpenVoiceOS/ovos-PHAL-plugin-mk1`

- New user config:
  ```json
  {
  "skills": {
    "blacklisted_skills": [
      "mycroft-joke.mycroftai"
    ]
  },
  "listener": {
    "silence_end": 0.5,
    "recording_timeout": 7,
    "VAD": {
      "module": "ovos-vad-plugin-silero"
    }
  },
  "hotwords": {
    "hey_mycroft": {
      "module": "ovos-ww-plugin-precise-lite",
      "model": "https://github.com/OpenVoiceOS/precise-lite-models/raw/master/wakewords/en/hey_mycroft.tflite",
      "listen": true,
      "sound": "snd/start_listening.wav",
      "expected_duration": 3,
      "trigger_level": 3,
      "sensitivity": 0.5,
      "fallback_ww": "hey_mycroft_openwakeword"
    },
    "hey_mycroft_openwakeword": {
      "module": "ovos-ww-plugin-openwakeword",
      "threshold": 0.4,
      "fallback_ww": "hey_mycroft_vosk"
    }
  },
  "tts": {
    "module": "ovos-tts-plugin-piper",
    "fallback_module": "ovos-tts-plugin-server",
    "ovos-tts-plugin-piper": {
      "voice": "alan-low"
    }
  },
  "padatious": {
    "regex_only": false
  },
  "g2p": {
    "module": "ovos-g2p-plugin-mimic"
  },
  "Audio": {
    "backends": {
      "OCP": {
        "playback_mode": 40,
        "preferred_audio_services": [
          "mplayer",
          "vlc"
        ],
        "autoplay": false,
        "min_timeout": 3,
        "max_timeout": 10
      }
    }
  },
  "logs": {
    "path": "/ramdisk/mycroft",
    "max_bytes": 2000000,
    "backup_count": 1
  },
  "system_unit": "imperial",
  "time_format": "half"
}
```
- `sudo systemctl enable avahi-daemon && sudo systemctl start avahi-daemon` (I like logging in via hostname)
- `pip install neon-mana-utils` (Mana!)
- ZRAM installation:
    - `sudo su`, then:
```bash
apt install zram-tools
sed -i '/#PERCENT=50/c\PERCENT=50' /etc/default/zramswap
cat <<EOF >> /etc/sysctl.conf

#zram
vm.vfs_cache_pressure=500
vm.swappiness=100
vm.dirty_background_ratio=1
vm.dirty_ratio=50
EOF
sysctl --system
systemctl enable zramswap
systemctl start zramswap
```
- Had to install mplayer from apt and then `pip install git+https://github.com/OpenVoiceOS/ovos-mplayer-plugin` to use mplayer with OCP (VLC doesn't seem to behave nicely all the time)
- `pip install git+https://github.com/OpenVoiceOS/ovos-vlc-plugin`
- `pip install git+https://github.com/OpenVoiceOS/ovos-ocp-youtube-plugin` (nice to have some kind of music playing capability out of the box)
- `osm install https://github.com/MycroftAI/skill-singing` (a classic Mark 1 skill)

Thanks for supporting the original hardware!

builderjer, 2023-09-03:
New image with many of the improvements here.  Some were not necessary, and some just not implemented with the image.

Can be tested here 
https://ovosimages.ziggyai.online/raspbian/mark1

builderjer, 2023-09-22:
@mikejgray did you test this image? can I close this?

mikejgray, 2023-09-22:
@builderjer I did, and there were some regressions that caused it to not work for me. Will try to get you more details tonight or this weekend