OpenVoiceOS/ZZZ-raspOVOS #90: Help understanding ovos-config

ByteRogue, 2023-09-08:
Hey all,

I just have a few questions to better understand ovos-config. How do I change the State/City and Timezone?

Also to a lesser extent, I can't seem to get Google TTS to work. I installed `pip install ovos-tts-plugin-google-tx`. Changed mycroft.conf to   
```
"tts": {
    "module": "ovos-tts-plugin-google-tx"
  }
```

But it still plays the default voice. I'm not sure what I'm doing wrong.

Thank you for all your help and this project!

Bulwagga, 2023-09-09:
Not sure about how to get Google TTS working, but currently ([Ovos-config](https://github.com/OpenVoiceOS/ovos-config) has an [issue](https://github.com/OpenVoiceOS/ovos-config/issues/59) right now that may be holding you back. 

Typing 

> ovos-config get -k /location/timezone/name

gets me my time zone name.  Trying to change it with 

> ovos-config set -k /location/timezone/name

gives me an error message.  For now I am creating a local mycroft.conf or editing the system copy.

Bulwagga, 2023-09-09:
If  you would rather have degrees in Fahrenheit rather then Celsius, you can use

> ovos-config set -k system_unit -v imperial

This edit works correctly, and creates (or edits an existing) ~/.config/mycroft/mycroft.conf.  After setting your new configuration, you'll need to restart OVOS to re-read your mycroft.conf files.  raspbian-ovos restart would be

> systemctl --user restart ovos.service

builderjer, 2023-09-09:
@ByteRogue, we at OVOS are happy you are testing our software. And I'm glad to hear that it works in a generic way. To answer your questions, I will redirect you to more specific help.

- location
  - what doesn't work? Is your time right? Or weather off, or???

- google-tx
  - honestly, I'm not sure how long this plugin will work without a personal Google api code. It is based on an old public code that can end at any time. 

For more support on both of these subjects I will redirect you to the [matrix chats](https://matrix.to/#/!QMaxYxzZZsZPCUgTZw:matrix.org?via=matrix.org&via=mozilla.org&via=kde.org) which are VERY  active with all of the developers

ByteRogue, 2023-09-10:
> @ByteRogue, we at OVOS are happy you are testing our software. And I'm glad to hear that it works in a generic way. To answer your questions, I will redirect you to more specific help.
> 
> * location
>   
>   * what doesn't work? Is your time right? Or weather off, or???
> * google-tx
>   
>   * honestly, I'm not sure how long this plugin will work without a personal Google api code. It is based on an old public code that can end at any time.
> * ovos-config
>   
>   * see the issue above and make a comment  there please.
> 
> For more support on all of these subjects I will redirect you to the [matrix chats](https://matrix.to/#/!QMaxYxzZZsZPCUgTZw:matrix.org?via=matrix.org&via=mozilla.org&via=kde.org) which are VERY active with all of the developers

Thank you for the response. 
location - what doesn't work? Is your time right? Or weather off, or???  --- 

> **Time is behind an hour. The location shown in the ovos-config show is a few states away. I'm unsure if that affects the outcome of what the weather command is, as, at least testing now, the temperature is correct but the humidity is way off. Though, even when specifying the weather in my state, the humidity outputs are the same—about 40% off**.

google-tx - honestly, I'm not sure how long this plugin will work without a personal Google api code. It is based on an old public code that can end at any time.   ---  

> **Understood. The only reason why I ask is it is still currently working with MyCroft.**

Another few questions. 
1} I dislike keep comparing to MyCroft but the response is immediate, with this, it's anywhere from 20-30 seconds. Is this a hardware limitation with the current optimization? (Pi 3b) 

2} Is there an option to view the live debug such as MyCroft's `mycroft-start debug`

Thanks!

ByteRogue, 2023-09-10:
> If you would rather have degrees in Fahrenheit rather then Celsius, you can use
>
>
> This edit works correctly, and creates (or edits an existing) ~/.config/mycroft/mycroft.conf. After setting your new [...]

Thank you! Worked perfectly.

builderjer, 2023-09-22:
Closing this.  Feel free to open another if needed

closed 2023-09-22 by builderjer