OpenVoiceOS/ZZZ-raspOVOS #91: [BUG] venv branch systemd ovos service failing

mikejgray, 2023-09-22:
```bash
mike@kubuntu:~/raspbian-ovos$ systemctl --user status ovos
× ovos.service - OVOS A.I. Software stack.
     Loaded: loaded (/home/mike/.config/systemd/user/ovos.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Fri 2023-09-22 17:20:01 CDT; 1min 34s ago
    Process: 9960 ExecStart=/home/mike/venv-ovos/bin/python3 /bin/true (code=exited, status=1/FAILURE)
   Main PID: 9960 (code=exited, status=1/FAILURE)
        CPU: 28ms

Sep 22 17:20:00 kubuntu systemd[933]: Starting OVOS A.I. Software stack....
Sep 22 17:20:01 kubuntu python3[9960]: SyntaxError: Non-UTF-8 code starting with '\xdb' in file /bin/true on line 2, but no encoding declared; see https://python.org/dev/peps/pep-0263/ for details
Sep 22 17:20:01 kubuntu systemd[933]: ovos.service: Main process exited, code=exited, status=1/FAILURE
Sep 22 17:20:01 kubuntu systemd[933]: ovos.service: Failed with result 'exit-code'.
Sep 22 17:20:01 kubuntu systemd[933]: Failed to start OVOS A.I. Software stack..
mike@kubuntu:~/raspbian-ovos$ ~/venv-ovos/bin/python3 /bin/true
SyntaxError: Non-UTF-8 code starting with '\xdb' in file /bin/true on line 2, but no encoding declared; see https://python.org/dev/peps/pep-0263/ for details
```

mikejgray, 2023-09-22:
Looks like at least part of this is due to the onnxruntime bug:

```bash
mike@kubuntu:~/raspbian-ovos$ grep ERROR ~/.local/state/mycroft/*.log
/home/mike/.local/state/mycroft/skills.log:2023-09-22 17:24:51.699 - skills - ovos_plugin_manager.utils:find_plugins:111 - ERROR - Failed to load plugin entry point EntryPoint(name='skill-alerts.neongeckocom', value='skill_alerts:AlertSkill', group='ovos.plugin.skill'): No module named 'mycroft_bus_client'
/home/mike/.local/state/mycroft/voice.log:2023-09-22 17:20:04.457 - voice - ovos_plugin_manager.vad:create:129 - ERROR - VAD plugin ovos-vad-plugin-silero could not be loaded!
```

mikejgray, 2023-09-22:
Per Daniel:

> so, after the fact you should be able to pip install onnxruntime~=1.15.0 to get the latest working version

builderjer, 2023-09-23:
This was a wip and I know it has bugs. If you narrowed it down to that, I will try and look this weekend.

mikejgray, 2023-09-23:
It actually turned out that instead of the OVOS service running /bin/true it was trying to run the venv Python command, followed by /bin/true. I didn't see where that was added to the service files so didn't update it in the PR. If you know offhand let me know so I can fix that in the PR.

builderjer, 2023-09-23:
might start installing the systemd files from here
https://github.com/OpenVoiceOS/ovos-systemd/tree/demycroftify

PR #92 (2023-09-23): fix: successful manual_user_install in kubuntu 22.04
Fix for:
breaking change in onnxruntime
missing neon-utils dependency
change in jokes skill location
removes local ww models that don't exist in base kubuntu installations
incorrect config key for piper plugin

Add:
easter egg skill
allow user to choose PulseAudio instead of PipeWire (fixes audio issue in kubuntu 22.04)

Closes #91
closed 2023-09-23 by mikejgray