OpenVoiceOS/ZZZ-raspOVOS #97: systemd processes don't start correctly

Bulwagga, 2023-10-28:
I started work on the uninstall script and I think I found the source of cosmOnaut's problems starting ovos from the manual build script. I'm not sure the *.service files are written correctly.  When I do a clean install, the ovos service fails to start because the command ExecStart=/home/ovos/venv-ovos/bin/python3 /bin/true tries to use the venv python3 exe as data. I will have a bit of a learning curve but I should be able to figure this out. It is a problem in by the manual install script.

austin-b, 2023-10-28:
I've just run across the same problem with a clean install after using the manual build script. My systemctl output is:
``Oct 28 17:22:56 systemd[1350]: Starting OVOS A.I. Software stack....
Oct 28 17:22:56 python3[16866]: SyntaxError: Non-UTF-8 code starting with '\xdb' in file /bin/true on line 2, but n>
Oct 28 17:22:56 systemd[1350]: ovos.service: Main process exited, code=exited, status=1/FAILURE
Oct 28 17:22:56 systemd[1350]: ovos.service: Failed with result 'exit-code'.
Oct 28 17:22:56 systemd[1350]: Failed to start OVOS A.I. Software stack..``

austin-b, 2023-10-28:
I seem to have fixed that error by changing the line to ``ExecStart=/home/tommyboi/ovos-src/venv-ovos/bin/python3 "$(/bin/true)"`` (after the script has run but before "Would you like to start?" prompt, and with a ``systemctl --user daemon-reload``) but now ovos-admin-phal.service is failing with 
```
× ovos-admin-phal.service - OVOS ADMIN PHAL
     Loaded: loaded (/etc/systemd/system/ovos-admin-phal.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Sat 2023-10-28 18:00:55 EDT; 6min ago
    Process: 25810 ExecStart=/home/tommyboi/ovos-src/venv-ovos/bin/python3 /home/ovos/.venv/bin/python /usr/libexec/ovos-systemd-admin-phal (code=exited, status=217/USER)
   Main PID: 25810 (code=exited, status=217/USER)
        CPU: 2ms

Oct 28 18:00:55 tommyboi-Swift-SF314-51 systemd[1]: ovos-admin-phal.service: Scheduled restart job, restart counter is at 4.
Oct 28 18:00:55 tommyboi-Swift-SF314-51 systemd[1]: Stopped OVOS ADMIN PHAL.
Oct 28 18:00:55 tommyboi-Swift-SF314-51 systemd[1]: ovos-admin-phal.service: Start request repeated too quickly.
Oct 28 18:00:55 tommyboi-Swift-SF314-51 systemd[1]: ovos-admin-phal.service: Failed with result 'exit-code'.
Oct 28 18:00:55 tommyboi-Swift-SF314-51 systemd[1]: Failed to start OVOS ADMIN PHAL.
Oct 28 18:02:26 tommyboi-Swift-SF314-51 systemd[1]: ovos-admin-phal.service: Start request repeated too quickly.
Oct 28 18:02:26 tommyboi-Swift-SF314-51 systemd[1]: ovos-admin-phal.service: Failed with result 'exit-code'.
Oct 28 18:02:26 tommyboi-Swift-SF314-51 systemd[1]: Failed to start OVOS ADMIN PHAL.
```

I'm extremely new to OVOS as this is my first time trying to install it but I'll see what I can find.

Bulwagga, 2023-10-29:
This is what I saw.  Currently rebuilding my environment so I'm catching up. Note that ovos-admin-phal.service is a system service not a user service. There may be some issue with permissions, ability to create temp files, etc. Thanks for the help!

Bulwagga, 2023-10-29:
Turns out none of the seven systemd services are starting.  If you're in a  hurry, this would be a good time to install one of the images. :)  If you've got some time, the ExecStart commands of all six user processes need to be edited... If I understand your environment correctly, the five files you haven't edited yet should be of the form
ExecStart= /home/tommyboi/ovos-src/venv-ovos/bin/python3  /home/tommyboi/ovos-src/venv-ovos/bin/ovos-systemd-messagebus etc.
Unfortunately this isn't enough to kick off the ovos-dinkum-listener.service and I'm looking into that next.
When those six are running I'll go after ovos-phal-admin.
Thanks for your patience.

austin-b, 2023-10-29:
I've noticed two more things in the service files that seem to be contributing to the problem, I think:
in ovos-admin-phal.service, in `ExecStart=/home/tommyboi/ovos-src/venv-ovos/bin/python3 /home/ovos/.venv/bin/python /usr/libexec/ovos-systemd-admin-phal`, the /home/ovos directory does not exist at all; and in the other user services there are  line `ExecStart=/home/tommyboi/ovos-src/venv-ovos/bin/python3 %h/.venv/bin/python /home/tommyboi/ovos-src/venv-ovos/bin/ovos-systemd-audio`, and %h/.venv does not exist (as far as I can find %H is supposed to be a systemd specifier for hostname, but I can't find anything for %h)

builderjer, 2023-10-29:
I posted this under the wrong issue 

https://github.com/OpenVoiceOS/raspbian-ovos/issues/94#issuecomment-1783978892

builderjer, 2023-10-29:
The `%h` is the user hone directory. Change the first part to where your venv is located

builderjer, 2023-10-29:
https://github.com/OpenVoiceOS/raspbian-ovos/blob/dev/manual_user_install.sh#L122

Start here changing to where your venv is

Bulwagga, 2023-10-29:
Got it. My local copy is now kicking off the user services just fine, but  ovos-dinkum-listener.service is dying and the log file is telling me it found but failed to load ovos_tts_plugin_mimic3 as the backup tts. I can replace it with something else to make it work, but I'm wondering why the local mycroft.conf doesn't override the global. My local backup tts is ovos-tts-plugin-mimic.
I'm concerned that my ovos process isn't reading my local mycroft.conf.

`2023-10-29 13:47:04.701 - audio - ovos_plugin_manager.tts:create:209 - INFO - Loaded plugin ovos-tts-plugin-server
2023-10-29 13:47:04.705 - audio - ovos_audio.service:_maybe_reload_tts:336 - INFO - (re)loading fallback TTS engine
2023-10-29 13:47:04.806 - audio - ovos_plugin_manager.tts:create:205 - INFO - Found plugin ovos-tts-plugin-mimic3-server
2023-10-29 13:47:04.813 - audio - ovos_tts_plugin_mimic3_server:__init__:44 - WARNING - please move to ovos-tts-server instead, this plugin is no longer maintained
 https://github.com/OpenVoiceOS/ovos-tts-server-plugin
2023-10-29 13:47:04.904 - audio - ovos_plugin_manager.tts:create:215 - ERROR - The TTS plugin "ovos-tts-plugin-mimic3-server" could not be loaded.
Available modules: ovos-tts-plugin-server,ovos-tts-plugin-dummy,ovos-tts-plugin-piper,ovos-tts-plugin-mimic,ovos-tts-plugin-mimic3-server
Traceback (most recent call last):
  File "/home/ovos/venv-ovos/lib/python3.9/site-packages/ovos_plugin_manager/tts.py", line 206, in create
    tts = clazz(lang=None,  # explicitly read lang from config
  File "/home/ovos/venv-ovos/lib/python3.9/site-packages/ovos_tts_plugin_mimic3_server/__init__.py", line 49, in __init__
    lang = lang.split("-")[0]
AttributeError: 'NoneType' object has no attribute 'split'
2023-10-29 13:47:04.909 - audio - ovos_audio.service:__init__:93 - ERROR - 'NoneType' object has no attribute 'split'
Traceback (most recent call last):
  File "/home/ovos/venv-ovos/lib/python3.9/site-packages/ovos_audio/service.py", line 90, in __init__
    self._maybe_reload_tts()
  File "/home/ovos/venv-ovos/lib/python3.9/site-packages/ovos_audio/service.py", line 337, in _maybe_reload_tts
    self._get_tts_fallback()
  File "/home/ovos/venv-ovos/lib/python3.9/site-packages/ovos_audio/service.py", line 365, in _get_tts_fallback
    self.fallback_tts = TTSFactory.create(cfg)
  File "/home/ovos/venv-ovos/lib/python3.9/site-packages/ovos_audio/tts.py", line 9, in create
    return OVOSTTSFactory.create(config)
  File "/home/ovos/venv-ovos/lib/python3.9/site-packages/ovos_plugin_manager/tts.py", line 206, in create
    tts = clazz(lang=None,  # explicitly read lang from config
  File "/home/ovos/venv-ovos/lib/python3.9/site-packages/ovos_tts_plugin_mimic3_server/__init__.py", line 49, in __init__
    lang = lang.split("-")[0]
AttributeError: 'NoneType' object has no attribute 'split'
2023-10-29 13:47:04.915 - audio - ovos_audio.service:on_error:44 - ERROR - Audio service failed to launch ('NoneType' object has no attribute 'split').
`

builderjer, 2023-10-29:
Check your config with `js`. Could be its not loading it

Bulwagga, 2023-10-29:
found it... local mycroft.conf aborting during load

Bulwagga, 2023-10-30:
@builderjer  I'm all set except for an issue loading tfltite... And I believe you're working on it. Should I issue pull request or should I wait? Thanks

builderjer, 2023-10-30:
How about this

https://github.com/OpenVoiceOS/raspbian-ovos/blob/dev/stage-audio/02-voice/02-run-chroot.sh#L6

builderjer, 2023-10-30:
Or is there another issue?

Bulwagga, 2023-10-30:
That fixed the tflite issue, but apparently there is a second problem loading the wake word:
`voice - ovos_plugin_manager.wakewords:load_module:172 - INFO - Loaded the Wake Word hey_mycroft with module ovos-ww-plugin-precise-lite
voice - ovos_plugin_manager.wakewords:create_hotword:196 - ERROR - Failed to load hotword: hey_mycroft - ovos-ww-plugin-precise-lite
voice - ovos_plugin_manager.wakewords:create_hotword:197 - ERROR - Model not found
Traceback (most recent call last):
  File "/home/ovos/venv-ovos/lib/python3.9/site-packages/ovos_plugin_manager/wakewords.py", line 194, in create_hotword
    return cls.load_module(module, hotword, ww_config, lang, loop)
  File "/home/ovos/venv-ovos/lib/python3.9/site-packages/ovos_plugin_manager/wakewords.py", line 173, in load_module
    return clazz(hotword, hotword_config, lang=lang)
  File "/home/ovos/venv-ovos/lib/python3.9/site-packages/ovos_ww_plugin_precise_lite/__init__.py", line 30, in __init__
    raise ValueError("Model not found")
ValueError: Model not found
`
Thoughts?

Bulwagga, 2023-10-30:
The model  '/home/ovos/.local/share/precise_lite/hey_mycroft.tflite' isn't getting pulled in from the new tflite location.

Bulwagga, 2023-10-31:
Making progress. Give me a day. I found the wakewords to download.

builderjer, 2023-10-31:
You can set the model url in the config.

https://github.com/OpenVoiceOS/ovos-ww-plugin-precise-lite#configuration

The "model" section can be a `http` url or a local path

Bulwagga, 2023-10-31:
Setting the URL did the trick.  I added a local copy to match the mycroft.conf and it didn't work.  Moving forward, TTS is pronouncing poorly.  Give me another day or two to track that down and I think we'll have a working system.

Bulwagga, 2023-11-01:
Existing local mycroft.conf gives me somewhat garbled speech:
`    "tts": {
        "module": "ovos-tts-plugin-server",
        "ovos-tts-plugin-piper": {
            "model": "alan-low"
            },
        "fallback_module": "ovos-tts-plugin-mimic",
        "ovos-tts-plugin-mimic": {
            "voice": "ap"
            }
        }
`

Rearranging the entry gives me speech I can recognize - Not sure what the difference is, and not sure how to implement into install script:
`    "tts": {
        "module": "ovos-tts-plugin-server",
        "ovos-tts-plugin-mimic": {
            "voice": "ap"
            },
        "fallback_module": "ovos-tts-plugin-mimic",
        "ovos-tts-plugin-piper": {
            "model": "alan-low"
            }
        }
`
Thoughts?

builderjer, 2023-11-01:
No idea how or why that would make a difference. The config is just a dictionary doesn't matter what the order is.

Bulwagga, 2023-11-01:
Syllables were getting left out... Working today.  I will work now to integrate with your changes, then a pull request

Bulwagga, 2023-11-02:
@austin-b does this work for you?

closed 2023-11-04 by Bulwagga

austin-b, 2023-11-05:
I've been able to get everything running with some changes to the service files and removing the comma from mycroft.conf that was causing a JSON read error, and finally got the model locations figured out.

Currently ovos is detecting my speech and trying to respond per the audio logs, but no audio is playing.

Bulwagga, 2023-11-05:
This is a great general interest question... I am going to open an issue where what we do will be more easily searchable.  Please join in there.

builderjer, 2023-11-05:
The new issue is [here](https://github.com/OpenVoiceOS/raspbian-ovos/issues/101)