OpenVoiceOS/ZZZ-raspOVOS #110: USB Audio Interface

unfirthman, 2023-12-10:
Hello! Thank you so much for this cool software! I would like to use a USB audio interface for routing audio input and outputs from my raspberry pi. Where can I alter settings to work on this?

Bulwagga, 2023-12-10:
Audio setup can take some tinkering.

I'm using pulseaudio with a Plantronics USB P3200 phone mic/speaker. It gets picked up automatically by OVOS. You might try alsamixer to set input/output. I have also used amixer and pactl.  Check out https://www.mankier.com/1/pactl for ideas to set default sound sources and sinks.

These two links may be helpful as well.
https://openvoiceos.github.io/community-docs/011-hardware_audio/
https://mycroft-ai.gitbook.io/docs/using-mycroft-ai/troubleshooting/audio-troubleshooting

unfirthman, 2024-02-09:
Thank you for this! Very helpful!

This can be closed now.