OpenVoiceOS/ZZZ-raspOVOS #117: ERROR - TTS synth failed! All OVOS TTS servers are down!

BigYellowHammer, 2024-02-21:
Hey,

I downloaded ovos-raspbian 2 days ago (19.02).
Tried to use fresh installation and looks like it can hear me, but cannot say anything.

audio.log

```log
2024-02-20 13:18:49.641 - audio - ovos_utils.signal:get_ipc_directory:26 - WARNING - Deprecation version=0.1.0. Caller=ovos_audio.__main__:26. Expected a dict config and got None.
2024-02-20 13:18:49.693 - audio - ovos_audio.service:__init__:48 - INFO - Starting Audio Service
2024-02-20 13:18:49.697 - audio - ovos_audio.service:on_started:31 - INFO - Audio service started.
2024-02-20 13:18:49.813 - audio - ovos_audio.service:_maybe_reload_tts:289 - INFO - (re)loading TTS engine
2024-02-20 13:18:49.818 - audio - ovos_bus_client.session:reset_default_session:531 - INFO - New Default Session Start: 17b2bd4b-f235-4220-a358-1cb252a68330
2024-02-20 13:18:50.133 - audio - ovos_utils.configuration:get_xdg_base - WARNING - Deprecation version=0.1.0. Caller=ovos_tts_plugin_mimic:93. configuration moved to the `ovos_config` package.
2024-02-20 13:18:50.205 - audio - ovos_utils.configuration:read_mycroft_config - WARNING - Deprecation version=0.1.0. Caller=ovos_tts_plugin_mimic:106. configuration moved to the `ovos_config` package.
2024-02-20 13:18:50.359 - audio - ovos_plugin_manager.tts:create:197 - INFO - Found plugin ovos-tts-plugin-server
2024-02-20 13:18:50.374 - audio - ovos_utils.file_utils:resolve_resource_file:152 - WARNING - Deprecation version=0.1.0. Caller=ovos_plugin_manager.templates.tts:526. Expected a dict config and got None.
2024-02-20 13:18:50.475 - audio - mycroft:<module>:37 - WARNING - mycroft has been deprecated! please start importing from ovos_core and companion packages
mycroft module remains available for backwards compatibility and will be removed in version 0.2.0
2024-02-20 13:18:56.376 - audio - ovos_plugin_manager.tts:create:200 - INFO - Loaded plugin ovos-tts-plugin-server
2024-02-20 13:18:56.382 - audio - ovos_audio.service:_maybe_reload_tts:308 - INFO - (re)loading fallback TTS engine
2024-02-20 13:18:56.459 - audio - ovos_plugin_manager.tts:create:197 - INFO - Found plugin ovos-tts-plugin-mimic
2024-02-20 13:18:56.463 - audio - ovos_utils.file_utils:resolve_resource_file:152 - WARNING - Deprecation version=0.1.0. Caller=ovos_plugin_manager.templates.tts:526. Expected a dict config and got None.
2024-02-20 13:18:56.565 - audio - ovos_utils.configuration:get_xdg_base - WARNING - Deprecation version=0.1.0. Caller=ovos_tts_plugin_mimic:93. configuration moved to the `ovos_config` package.
2024-02-20 13:18:56.570 - audio - ovos_utils.configuration:read_mycroft_config - WARNING - Deprecation version=0.1.0. Caller=ovos_tts_plugin_mimic:106. configuration moved to the `ovos_config` package.
2024-02-20 13:18:56.572 - audio - ovos_plugin_manager.tts:create:200 - INFO - Loaded plugin ovos-tts-plugin-mimic
2024-02-20 13:18:56.637 - audio - ovos_plugin_manager.audio:load_audio_service_plugins:84 - INFO - Loading audio service plugin: ovos_common_play
2024-02-20 13:18:58.730 - audio - ovos_audio.service:execute_tts:320 - INFO - Speak: I have created a temporary WiFi network called OVOS. Connect to it from your phone or computer and go to start dot openvoiceos dot com.
2024-02-20 13:18:58.746 - audio - ovos_audio.service:execute_tts:326 - ERROR - TTS synth failed! All OVOS TTS servers are down!
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/ovos_audio/service.py", line 323, in execute_tts
    self.tts.execute(utterance, ident, listen,
  File "/usr/local/lib/python3.9/dist-packages/ovos_plugin_manager/templates/tts.py", line 724, in execute
    self._execute(sentence, ident, listen, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/ovos_plugin_manager/templates/tts.py", line 750, in _execute
    audio_file, phonemes = self.synth(sentence, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/ovos_plugin_manager/templates/tts.py", line 810, in synth
    audio.path, phonemes = self.get_tts(sentence, str(audio), **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/ovos_tts_plugin_server/__init__.py", line 31, in get_tts
    data = self._get_from_servers(params, sentence, servers)
  File "/usr/local/lib/python3.9/dist-packages/ovos_tts_plugin_server/__init__.py", line 45, in _get_from_servers
    raise RemoteTTSException(f"All OVOS TTS servers are down!")
ovos_plugin_manager.templates.tts.RemoteTTSException: All OVOS TTS servers are down!
2024-02-20 13:19:00.420 - ovos.common_play - WARNING - Tried to initialize ovos.common_play multiple times, ignoring
2024-02-20 13:19:01.357 - ovos.common_play - WARNING - Tried to initialize ovos.common_play multiple times, ignoring
2024-02-20 13:19:02.026 - audio - ovos_plugin_manager.audio:load_audio_service_plugins:84 - INFO - Loading audio service plugin: ovos_audio_simple
2024-02-20 13:19:02.029 - audio - ovos_audio.audio:load_services:75 - INFO - OCP - OVOS Common Play set as default backend
2024-02-20 13:19:02.033 - audio - ovos_audio.service:on_alive:27 - INFO - Audio service is alive.
2024-02-20 13:19:12.079 - audio - ovos_plugin_common_play.ocp:register_ocp_intents:120 - INFO - OCP intents missing, registering for <ovos_plugin_common_play.ocp.OCP object at 0x7f822e4520>
```

Here is my OVOS-config return for tts

```log
┤
│ tts                                                          │                                                                                                            │
│     pulse_duck                                               │ False                                                                                                      │
│     module                                                   │ ovos-tts-plugin-server                                                                                     │
│     fallback_module                                          │ ovos-tts-plugin-mimic                                                                                      │
│     ovos-tts-plugin-mimic                                    │                                                                                                            │
│         voice                                                │ ap                                                                                                         │
│     ovos-tts-plugin-mimic3-server                            │                                                                                                            │
│         voice                                                │ en_UK/apope_low      
```

Any idea what might be wrong and how to fix it?

I was previously using Mycroft on my raspberry pi, but recently everything got turned off so decided to move to something new.
I love that project and hoping I can contribute to it.

Thanks for all the hard work on it!

builderjer, 2024-02-21:
Use alsamixer and check your volume. Even if the public servers are down, it should fallback to a local mimic

BigYellowHammer, 2024-02-27:
Mystery solved!
I used images from openvoiceos.org site:
https://downloads.openvoiceos.org/images/raspbian/

While the correct one is in documentation:
https://ovosimages.ziggyai.online/raspbian/newest/

Installing new one solved all the problems and everything works like a charm.
Is official openvoiceos.org site content somewhere on github? I'm happy to issue PR that fixes those links so that everybody in the future will use correct images.

builderjer, 2024-02-27:
I am glad this worked for you. The openvoiceos.org site is in the process of being updated to something that can be maintained better.  Sample code is at https://github.com/OpenVoiceOS/ovos-org-website

And a sample of it running at https://ovosorg.ziggyai.online/home/

This is still a WIP and not yet released from the org. But pr's are always welcome

BigYellowHammer, 2024-02-28:
I see this site is having all the correct links. So its only openvoiceos.org having problems.
Anyway - thanks for all your help

closed 2024-02-28 by BigYellowHammer