OpenVoiceOS/ZZZ-raspOVOS #119: Cannot change TTS

BigYellowHammer, 2024-02-28:
Hey,

Im trying to change voice for my raspOVOS.
I modified /etc/mycroft/mycroft.conf to be following

```json
{
    "enable_old_audioservice": false,
    "logs": {
        "path": "/ramdisk/mycroft",
        "max_bytes": 1500000,
        "backup_count": 1
    },
    "listener": {
        "remove_silence": true
    },
    "hotwords": {
        "hey_mycroft": {
            "model": "/home/ovos/.local/share/precise_lite/hey_mycroft.tflite"
        },
        "hey_mycroft_vosk": {
            "model_folder": "/home/ovos/.local/share/vosk/vosk-model-small-en-us-0.15"
        },
        "hey_ziggy": {
            "module": "ovos-ww-plugin-vosk",
            "debug": false,
            "rule": "fuzzy",
            "lang": "en",
            "samples": [
                "hey ziggy",
                "hay ziggy"
                ],
            "threshold": 0.8
        }
    },
    "tts": {
        "module": "ovos-tts-plugin-piper",
        "ovos-tts-plugin-server": {
            "sentence_tokenize": true
        },
        "ovos-tts-plugin-piper": {
            "model": "amy-low",
            "sentence_tokenize": true
        },
        "fallback_module": "ovos-tts-plugin-mimic",
        "ovos-tts-plugin-mimic": {
            "voice": "ap"
        }
    }
}
```

I'm expecting it will load piper plugin and use model **amy-low**, but im still hearing default **alan-low** voice.
Is that something that doesn't work yet?

builderjer, 2024-03-04:
First, you should not edit `/etc/mycroft/mycroft.conf`.  All changes will be overwritten when I finally enable OTA updates.
All user modifications should go into `~/.config/mycroft/mycroft.conf`.

Now the issue

`ovos-media` is new, and not completely supported in raspOVOS by default, so unless you explicitly installed it, this config will not work.

Remove the  `"enable_old_audioservice": false,` and try again

BigYellowHammer, 2024-03-04:
No joy :/
Still hear the same voice.

**/etc/mycroft/mycroft.conf** looks like this now

```json

{
    "logs": {
        "path": "/ramdisk/mycroft",
        "max_bytes": 1500000,
        "backup_count": 1
    },
    "listener": {
        "remove_silence": true
    },
    "hotwords": {
        "hey_mycroft": {
            "model": "/home/ovos/.local/share/precise_lite/hey_mycroft.tflite"
        },
        "hey_mycroft_vosk": {
            "model_folder": "/home/ovos/.local/share/vosk/vosk-model-small-en-us-0.15"
        },
        "hey_ziggy": {
            "module": "ovos-ww-plugin-vosk",
            "debug": false,
            "rule": "fuzzy",
            "lang": "en",
            "samples": [
                "hey ziggy",
                "hay ziggy"
                ],
            "threshold": 0.8
        }
    },
    "tts": {
        "module": "ovos-tts-plugin-server",
        "ovos-tts-plugin-server": {
            "sentence_tokenize": true
        },
        "ovos-tts-plugin-piper": {
            "model": "amy-low",
            "sentence_tokenize": true
        },
        "fallback_module": "ovos-tts-plugin-mimic",
        "ovos-tts-plugin-mimic": {
            "voice": "ap"
        }
    }
}
```

**~/.config/mycroft/mycroft.conf**

```json

{
    "log_level": "INFO",
    "tts": {
        "module": "ovos-tts-plugin-piper",
        "ovos-tts-plugin-server": {
            "sentence_tokenize": true
        },
        "ovos-tts-plugin-piper": {
            "model": "amy-low",
            "sentence_tokenize": true
        },
        "fallback_module": "ovos-tts-plugin-mimic",
        "ovos-tts-plugin-mimic": {
            "voice": "ap"
        }
    }
}
```

**/ramdisk/mycroft/audio.log**

```log
2024-03-04 06:57:20.860 - audio - ovos_audio.service:__init__:57 - INFO - Starting Audio Service
2024-03-04 06:57:20.865 - audio - ovos_audio.service:on_started:39 - INFO - TTS service started.
2024-03-04 06:57:23.382 - audio - ovos_audio.service:_maybe_reload_tts:312 - INFO - (re)loading TTS engine
2024-03-04 06:57:23.895 - audio - ovos_audio.playback:run:197 - INFO - PlaybackThread started
2024-03-04 06:57:24.837 - audio - ovos_plugin_manager.tts:create:205 - INFO - Found plugin ovos-tts-plugin-piper
2024-03-04 06:57:25.033 - audio - mycroft:<module>:37 - WARNING - mycroft has been deprecated! please start importing from ovos_core and companion packages
mycroft module remains available for backwards compatibility and will be removed in version 0.2.0
2024-03-04 06:57:25.148 - audio - ovos_plugin_manager.utils.tts_cache:load_persistent_cache:263 - INFO - Persistent TTS cache files loaded successfully.
2024-03-04 06:58:06.186 - audio - ovos_plugin_manager.tts:create:209 - INFO - Loaded plugin ovos-tts-plugin-piper
2024-03-04 06:58:06.189 - audio - ovos_audio.service:_maybe_reload_tts:332 - INFO - (re)loading fallback TTS engine
2024-03-04 06:58:06.243 - audio - ovos_plugin_manager.tts:create:205 - INFO - Found plugin ovos-tts-plugin-mimic
2024-03-04 06:58:06.297 - audio - ovos_plugin_manager.utils.tts_cache:load_persistent_cache:263 - INFO - Persistent TTS cache files loaded successfully.
2024-03-04 06:58:06.384 - audio - ovos_plugin_manager.tts:create:209 - INFO - Loaded plugin ovos-tts-plugin-mimic
2024-03-04 06:58:08.268 - audio - ovos_audio.audio:find_default:72 - INFO - Finding default audio backend...
2024-03-04 06:58:08.270 - audio - ovos_audio.audio:find_default:80 - INFO - no default found
2024-03-04 06:58:08.273 - audio - ovos_audio.service:on_alive:35 - INFO - TTS service is alive.
2024-03-04 06:58:08.274 - audio - ovos_audio.service:run:243 - WARNING - audio service has moved to ovos-media, if you already migrated to ovos-media set "enable_old_audioservice": false in mycroft.conf
2024-03-04 06:58:08.378 - audio - ovos_audio.service:run:247 - WARNING - No audio backends loaded! Audio playback is not available
2024-03-04 06:58:08.379 - audio - ovos_audio.service:run:249 - INFO - Running audio service in TTS only mode
2024-03-04 06:59:00.335 - audio - ovos_audio.service:execute_tts:344 - INFO - Speak: I am doing well, thank you
2024-03-04 06:59:01.416 - audio - ovos_plugin_manager.utils.tts_cache:load_persistent_cache:263 - INFO - Persistent TTS cache files loaded successfully.

```

Will try to enable some debug mode to see what TTS is actually getting used as I don't see if from the normal logs.

builderjer, 2024-03-04:
If you are running locally, change
`"model": "amy-low"`  to  `"voice": "amy-low"`
and check your results

BigYellowHammer, 2024-03-04:
That worked! It even automatically downloads other voices

Thanks a lot for your help on that

closed 2024-03-04 by BigYellowHammer